RedBearLab/nRF51822-Arduino #12: nRF51822-Arduino doesn't support s120 or s130

joshuajnoble, 2015-02-09:
Hi, s120 is already production ready and s130 is very close, it would be great if this library could support it :)

I'd be happy to help to try to support s120 if you could open up some of the toolchain.

closed 2015-02-15 by joshuajnoble

kteza1, 2015-04-23:
We need s120. Please support it.

luposlip, 2015-04-30:
@joshuajnoble you've closed this - did you find a way around getting s120 and s130 to work?

Cheong2K, 2015-05-05:
Hi @luposlip,

Use Keil and Nordic SDK to do that, after compiled a hex file, you can use drag and drop method to load to the BLE board.

luposlip, 2015-05-05:
Thanks for the input @Cheong2K. Still looking forward to your support with Arduino SDK though! ;-)

luposlip, 2015-05-05:
I see that Keil only works on Windows. Any alternatives for OSX?