RedBearLab/nRF51822-Arduino #15: Blink example is not working for the BLE Nano through the Arduino IDE

samolds, 2015-03-06:
I am using OS X 10.9.5.
I am using Arduino 1.5.8.
I cloned this repo and ran

    cp ...path/nRF51822-Arduino/arduino-1.5.x/hardware/arduino/RBL_nRF51822 /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/.

In the Arduino IDE I did Sketch > Import Library > Add Library and added the directory in ...path/nRF51822-Arduino/arduino-1.5.x/hardware/arduino/RBL_nRF51822/nrf51822_SDK

I dragged the bootloader in ...path/nRF51822-Arduino/bootloader/bootloader.hex to the MBED drive.

In the Arduino IDE I selected Tools > Board > RBL nRF51822 and then I selected Tools > Port > /dev/tty.usbmodem1412 (COM66 isn't an option)
At this point I figure I have followed all of the directions in the README exactly. So I opened up the Blink Example in the Arduino IDE and clicked upload. The LED on the MK20 starts flashing to signal that it is receiving the data. The Arduino IDE says "Done uploading" and then outputs:

    Sketch uses 4,408 bytes (2%) of program storage space. Maximum is 196,608 bytes.
    Found programmer: Id = "RBL-NRF"; type = S
    Programmer supports buffered memory access with buffersize=256 bytes.

and the LED on the BLE Nano board doesn't start blinking.

I know that my board is functioning correctly because I followed the tutorial to get the LED blinking through the mbed online method and I got it to work. Did I miss a step or do something incorrectly? Any help is appreciated.

Cheong2K, 2015-03-09:
Hi,

Please select BLE Nano board rather than the RBL_nRF51822 board.

samolds, 2015-03-10:
Thank you for the reply. I am pretty confident I tried that at least once, but that appears to have solved it. The README currently says to select the RBL_nRF51822 board, not BLE Nano. Can I submit a Pull Request, or is that a small enough change that you'd just like to make the update?

Anyway, thank you so much.

Cheong2K, 2015-03-11:
Thanks for pointing out the README, we will fix it soon.

closed 2015-03-11 by Cheong2K

jtpack, 2015-05-10:
Hello, uploading blink sketch to the BLE Nano in Arduino IDE (I am using 1.6.2) works - LED blinks properly. However, I see the message "Found programmer: Id = "RBL-NRF"; type = S Programmer supports buffered memory access with buffersize=256 bytes." every time I upload . Is this normal?

[image]

bogerssja, 2015-09-01:
@jtpack same issue here. did you fix it?

benfberkowitz, 2016-07-24:
Is there any movement on this error:

Found programmer: Id = "RBL-nRF"; type = S
Programmer supports buffered memory access with buffersize=256 bytes.

I tried reinstalling packages, and changing the Arduino programmer, but have not made any progress