RedBearLab/nRF51822-Arduino #18: s130 support

kteza1, 2015-04-14:
Are you going to add s130 support in near future?

luposlip, 2015-04-30:
:+1: I'm eagerly awaiting s130 support too - nordic semiconductors will release their 1.0 SDK to support this the 15th of May 2015 - latest.

luposlip, 2015-05-04:
@Cheong2K, do you have any idea if it'll take you 1 week, 1 month or 1 year to finish s130 support in you Arduino implementation? This is really important for me, if we'd opt for the your Nano board or the raw nRF51822 chip for our product. Thanks.

Cheong2K, 2015-05-04:
Nordic has released the S130 SoftDevice production version, we are working on it, hopefully, we can release it within this month but a lot of testing to do.

Our Arduino library is for fast prototyping and demonstration only, for your own products, we highly suggest you to use Nordic SDK directly and get support from them, since they have a team to support you but RedBearLab has limited resource only. Also, please consider to contribute your work.

Thank you for choosing the BLE Nano and we will try our best to do it well

luposlip, 2015-05-04:
Alright, thanks a lot for your input!
If there is anything I can do to help testing etc., please let me know.

rogerclarkmelbourne, 2015-10-31:
It looks like the bootloader now uses the S130 softdevice, (as does the code)

Can you confirm this, and also what version of the SDK it relates to, i.e is this now based on Version 9.0.x of the SDK?