RedBearLab/nRF51822-Arduino #22: OTA bootloader

adrianocos, 2015-05-27:
Hello, in the directory of the repository I see the bootloader files: bootloader_Avrdude_OTA_HCI_20150422.hex

how I do the PC side or Android or iOS to use the functionality OTA?

I see that the OTA is in functionality to do list, it is already ready?

Cheong2K, 2015-06-29:
Yes, bootloader_Avrdude_OTA_HCI_20150422.hex is a S110 bootloader for SDK 6.x.
You can use Nordic's nRF ToolBox to try.
We are working on S130 bootloader now.

chenchenece, 2017-10-18:
Hello @Cheong2K, we are currently using this toolchain to develop a MDBT40 based application. However we meet a lot issues on merging bootloader.hex (downloaded from this github) and Arduino exported application. 

Could you please show me how to merge bootloader.hex and Arduino exported code into single .hex file?

Or can you please indicate where I can download the bootloader hex file without Softdevice being merged, it will be best if you can show us the bootloader source code??

Thanks for the help!