RedBearLab/nRF51822-Arduino #33: Can this software be used with other 51822 modules?

rkertesz, 2015-10-18:
Long ago, rfdigital released their rfduino and tried to open source much of their work but many aspects of the Nordic stack were simply binary blobs. Since then we've seen the chip incorporated into models sold through multiple design houses and stores, i.e. redbear, /semiconductor store, numerous Chinese outfits, and even tindie (ucXpresso lily pad). 
Is there a way we can still write software in wiring/arduino and flash the hardware on these platforms or is this software specific to RBL's devices due to issues similar to what rfdigital ran into? Their silicon couldn't be reflashed without losing arduino programming capability. 
Apologies if this is a sophomore question.

rogerclarkmelbourne, 2015-10-29:
I can't answer all of this.

But I used OpenOCD and a STLink (SWD) programmer to flash the bootloader.hex to a nRF51822 module I bought from eBay

The module now appears on Bluetooth as a DFU device, but I can't see a way to upload to this DFU device from a PC 

I suppose I could try compiling a blank sketch and flashing it to Flash bank 1 (i.e of 0 and 1) or perhaps it needs to be flashed to a specific address??

rogerclarkmelbourne, 2015-10-30:
It does appear to be possible to use this with other modules, but first you need to flash the bootloader onto the module.

I already had a Cortex M0 compatible programmer (a STLink for STM32 processors) which could be used with OpenOCD to flash the bootloader.

Once the bootloader had been flashed it advertises over BLE as a DFU device and its possible to upload an "Application" using the nRF51 toolbox, as described in the OTA section of the readme

However I have a feeling there is an easier way, i.e just flash using the SWD programmer (STLink) all the time to flash

But, I"m still trying to work out if the nRF toolbox app merges a Softdevice with the hex file before uploading it to via DFU to the module (I suspect it may, even though it looks like the bootloader also mainly consists of a Softdevice)

rogerclarkmelbourne, 2015-10-31:
Definitive answer is yes.

It does work on other boards.

However you need a SWD programmer to be able to use it.
I use a STLink adaptor, but JLink is also known to work, and I suspect any SWD programmer that is supported by OpenOCD and works with Cortex-M0 will work

See this thread 

http://stm32duino.com/viewtopic.php?f=45&t=682

shlomozippel, 2016-03-22:
I've had success getting this working on other nRF51 boards, specifically the PTR5518 and PTR5528 modules: https://github.com/ChromaTech/nRF51822-Arduino

There are 2 things to look out for when using on a different nRF module:
- External crystal vs. internal oscillator. This changes the init / softdevice init code. The codebase actually takes this into account, look for reference to TARGET_HRM1017 or TARGET_DELTA_DFCM_NNN40, those are 2 nRF51 modules without an external crystal.
- Pin mappings. This mainly involves using the Arduino IDE "Variants" but also required some fiddling around with analog pin mappings, which you can see in the link above to my chroma fork.

As @rogerclarkmelbourne mentioned you can use any SWD programmer. I use the JLink that is built into the nRF51 dev kit after exporting the hex from arduino IDE.

rogerclarkmelbourne, 2016-03-22:
@shlomozippel 

Thanks for the link to your repo

Unfortunately I've not had time to work on my repo recently, or to get it to work with the Wireless-Tag modules I have (the repo currently only works with devices with 32khz crystal)

I can build for the Wireless-Tag module (without 32khz crystal) using mbed; so I'd always presumed I would be able to copy the target files for into my repo.

But I simply don't have time to investigate this at the moment (as I'm working on 2 concurrent contracts which take over 100% of my normal allotted worktime :-( )