RedBearLab/nRF51822-Arduino #35: Why do you use Serial1 as the main serial device

rogerclarkmelbourne, 2015-10-31:
Hi
I wondered why you used Serial1 for the serial debugging etc

Normally on the Arduino platform, Serial is used for the default debugging / serial comms port.

It seems to be possible to use an instance of Serial to call  Serial.begin()

However, Serial.println() fails to compile

The error I get is

Blink.ino: In function 'void setup()':
Blink.ino:21:9: error: expected unqualified-id before '.' token
expected unqualified-id before '.' token

So, its like something is defining "Serial" but perhaps its part of the BLE HAL or a macro, as it gives that strange error when I compile.

rogerclarkmelbourne, 2015-10-31:
Sorry to probably answer my own question, but this appears to be because the mbed files you are using have a class called Serial.

Cheong2K, 2015-11-02:
Hi,

Yes, we will fix this as soon as possible, it is not good at all.

--Cheong

sandeepmistry, 2015-11-02:
Here's a workaround I've been using for now: https://github.com/sandeepmistry/nRF51822-Arduino/commit/ab5f33b81777c78d412142567b1dc65d97893feb

rogerclarkmelbourne, 2015-11-03:
Thanks Sandeep

We meet again...  but this time not for ESP8266 ;-)

I'll probably just use your repo if you have already fixed this. And use your S130 branch.
Or I may just update my local copy.
I've moved it into the Arduino/hardware folder as it makes modification easier.

BTW. What do you flash with.I'm using an STLink (as I already have that), and I have a JLink clone but I can't get it to work :-(

sandeepmistry, 2015-11-03:
@rogerclarkmelbourne 

> BTW. What do you flash with.I'm using an STLink (as I already have that), and I have a JLink clone but I can't get it to work :-(

I'm using the MK20 USB board that comes with the [BLE Nano](http://redbearlab.com/blenano/) or [nRF51822 board](http://redbearlab.com/redbearlab-nrf51822/) which has a built-in MK20.

rogerclarkmelbourne, 2015-11-04:
Thanks Sandeep

I'm actually using a standalone nRF51822 module board, that doesn't have the freesscale (MK20), so I'm using an external SWD and external serial.
I'm hoping to combine them both by using a device called a Black Magic Probe, (which you can flash onto a $5 generic board from eBay)
But I need to sort out the upload control for that before I can use it.

But I think it basically would do roughly what the MK20 does except it uses GDB to upload rather than AVRDude

Anyway, thanks again for the info on Serial

shirish47, 2016-04-24:
Hi does serial communication on NRF51 Require pins other than TX,RX, GND? because I have setup my hardware as @rogerclarkmelbourne has shown in blog my NRF51822 is working and I can program it bt I am not able to communicate over UART I tried examples for UART.. p0.11 pin goes to pin 2 of Adafruit UART cable. and P0.09 to to 3rd pin of FTDI cable GND is last pin.  
[image]

rogerclarkmelbourne, 2016-04-24:
I didnt change the pin mapping for the UART. So it uses the same pins as the original RBL code (and RBL hardware).

I was going to remap the pins to make them easier to use for other boards, e.g. like the Waveshare module and motherboard.
However I found that WaveShare also use the same pins for the UART, ( p 9 and p 11)

So it seemed best to leave Serial on the same pins.

The only difference in the Serial is that I used to patch from Sandeep, so that it use Serial instead of Serial1 for the hardware serial port, as this is more normal on Arduino systems.

shirish47, 2016-04-25:
do you use P0.10 and P0.08 (RTS & CTS )in for usart?  Unfortunately I am not sure why I am not able to get Serial working wiring is proper.  :( let me debug more. without serial I can't do much. :(

rogerclarkmelbourne, 2016-04-25:
Take a look in

https://github.com/rogerclarkmelbourne/nRF51822-Arduino/blob/S130/arduino-1.6.x/hardware/RBL/RBL_nRF51822/cores/RBL_nRF51822/pin_transform.cpp

		case 0 : return_pin = P0_11;break;//D0/RX
		case 1 : return_pin = P0_9;break;//D1/TX

I'm sure Serial works OK on those pins, as I use it myself.

I don't use RTS or CTS, both of those pins is not connected on my WaveShare board

http://www.waveshare.com/nrf51822-eval-kit.htm, and it works fine

http://www.waveshare.com/w/upload/1/1b/BLE400-Schematic.pdf

shirish47, 2016-04-25:
@rogerclarkmelbourne  Ok I think its with the FTDI cable, voltage level problems. .:) just forgot I am working with 3 volts range. ;) btw found serial pin on BMP stm32 board.. we need to properly document pins of some of the STM boards.. I was following one of the image I found on internet bt its not exactly to my board.. anyways we can take that matter on STM32 page. thanks for the schematics. :)

rogerclarkmelbourne, 2016-04-25:
I normally use a Prolific USB to Serial, which seems to work fine on 3.3V logic.

One thing with the BMP is that the serial speed is fixed, so you have to use the same speed with the nRF51822, I can't remember what the speed is, but I suspect its 9600.)

shirish47, 2016-04-25:
yes its 9600.. :)

rogerclarkmelbourne, 2016-04-25:
BTW. Are you using my fork of RBL's code or the original (i.e in this repo).?

shirish47, 2016-04-25:
yours ..

rogerclarkmelbourne, 2016-04-25:
OK. What confused me is that this issue is on RBL's original repo not my fork.

Anyway,,, No worries. ;-)