RedBearLab/nRF51822-Arduino #42: arduino-builder in 1.6.6 doesn't pick up NRF51 and other expected macros

george-hawkins, 2015-12-06:
I know the 1.6.5 Arduino IDE is the currently recommended version for nRF51822-Arduino - this is a heads up and explanation of a breaking change that seems to have been introduced in 1.6.6.

If your `.ino` file (or a header it includes) depends on the `NRF51` macro (or any of the other normally defined macros) then it will break with 1.6.6.

This is because `arduino-builder` (that converts a `.ino` into a `.cpp`) uses a different recipe to the normal `cpp` recipe and this recipe does not currently pick up all the additional defines like `NRF51`.

Here's how to see the source of the problem - if you download and unpack [arduino-1.6.6-linux64.tar.xz](https://www.arduino.cc/download_handler.php?f=/arduino-1.6.6-linux64.tar.xz) you'll find `arduino-1.6.6/hardware/platform.txt` - open it in an editor and you'll see the lines:

```
preproc.includes.flags=-w -x c++ -M -MG -MP
recipe.preproc.includes="{compiler.path}{compiler.cpp.cmd}" {preproc.includes.flags} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.cpp.extra_flags} {build.extra_flags} {includes} "{source_file}"
```

The crucial thing is that it doesn't include all the defines (such as `-DNRF51`) then we see in `compiler.cpp.flags` in [`nRF51822-Arduino/arduino-1.6.x/hardware/RBL/RBL_nRF51822/platform.txt:21`](https://github.com/RedBearLab/nRF51822-Arduino/blob/c6d76697cb8f6df69315155c194068b911f8d923/arduino-1.6.x/hardware/RBL/RBL_nRF51822/platform.txt#L21), i.e.:
```
compiler.cpp.flags=-c -g -Os -w -std=gnu++98 -fno-common -fmessage-length=0 -Wall -fno-exceptions -ffunction-sections -fdata-sections -fomit-frame-pointer -nostdlib --param max-inline-insns-single=500 -fno-rtti -DBLE_STACK_SUPPORT_REQD -DDEBUG_NRF_USER -DNRF51 -DTARGET_NRF51822 -DTARGET_M0 -DTARGET_CORTEX_M -DTARGET_NORDIC -DTARGET_NRF51822_MKIT -DTARGET_MCU_NRF51822 -DTOOLCHAIN_GCC_ARM -DTOOLCHAIN_GCC -D__CORTEX_M0 -DARM_MATH_CM0 -DMBED_BUILD_TIMESTAMP=1435023129.97 -D__MBED__=1 
```

If you scroll over above you'll see that `recipe.preproc.includes` expects to pick up `compiler.cpp.extra_flags` rather than `compiler.cpp.flags` as defined in `RBL_nRF51822/platform.txt`.

Note: oddly it's not clear where the file `arduino-1.6.6/hardware/platform.txt` comes from - if I clone the [Arduino code base](https://github.com/arduino/Arduino/) and checkout revision [`0992606`](https://github.com/arduino/Arduino/commit/0992606) that's supposed to correspond to 1.6.6 I can't find any equivalent file.

george-hawkins, 2015-12-06:
**Solved:** I got things to work with the latest version of the Arduino IDE - i.e. 1.6.6.

The Arduino people have been fiddling about a lot with `platform.txt` in recent IDE releases and the [`nRF51822-Arduino/.../platform.txt`](https://github.com/RedBearLab/nRF51822-Arduino/blob/S130/arduino-1.6.x/hardware/RBL/RBL_nRF51822/platform.txt) has drifted apart a bit from the layouts now used in the [Arduino repository](https://github.com/arduino/Arduino/).

So I rearranged [`nRF51822-Arduino/.../platform.txt`](https://github.com/RedBearLab/nRF51822-Arduino/blob/S130/arduino-1.6.x/hardware/RBL/RBL_nRF51822/platform.txt) to match as closely as possible the layout of the Arduino [SAM `platform.txt`](https://github.com/arduino/Arduino/blob/master/hardware/arduino/sam/platform.txt) (except for the Avrdude section which I rearranged to match up as much as possible with the Avrdude section if the [AVR `platform.txt`](https://github.com/arduino/Arduino/blob/802c7456909f7b06780ffaa1f5acb379a9e337e4/hardware/arduino/avr/platform.txt#L91)).

I did all this while keeping all the RBL specific behavior (and removed some elements that are no longer required in the newer IDE releases).

The only really important thing was to separate out the RBL specific #defines like `NRF51` into the properties `compiler.c.extra_flags` and `compiler.cpp.extra_flags` so that the IDE's `.ino` converter can pick up `compiler.cpp.extra_flags`.

Other than doing this the only real addition that I made was to add the `-DNRF51_S130` define that @sandeepmistry already proposed in issue #41 so that one knows that the S130 soft device is being used.

Here is the [new `platform.txt`](https://github.com/RedBearLab/nRF51822-Arduino/files/53297/platform.txt) that will work with version 1.6.6 of the Arduino IDE. Despite looking very different, it is equivalent to [`nRF51822-Arduino/.../platform.txt` at revision `c6d7669`](https://github.com/RedBearLab/nRF51822-Arduino/blob/c6d76697cb8f6df69315155c194068b911f8d923/arduino-1.6.x/hardware/RBL/RBL_nRF51822/platform.txt), i.e. the latest version existing on Dec 6, 2015.

PS best of luck with the [DUO Kickstarter](https://www.kickstarter.com/projects/1991736672/redbear-duo-a-small-and-powerful-wi-fi-ble-iot-boa) - I'm looking forward to receiving mine :smile:

Cheong2K, 2015-12-07:
@george-hawkins,
Thank you for your good work and support, we will do more better with the open source world!
 --Cheong