RedBearLab/nRF51822-Arduino #45: Bootloader.hex too big (RBL Nano)

sk1tt1sh, 2015-12-31:
I'm trying to follow the arduino instructions. I have flashed the RBL driver to the mk20 usb device (hold button plug in drag RBL_nRF51822_MK20.bin to it) and now I am trying to put bootloader.hex on it for arduino programming and it says "File is too large".

When attmpting to upload from arduino ide it doesn't connect: "Connecting to programmer: .avrdude: butterfly_recv(): programmer is not responding" I have installed the CDC serial driver and selected the correct com port.

How do I tell which version of RBL Nano I have? The Board manager show 1.0 and 1.5

Thanks for your help!

Cheong2K, 2016-01-10:
Download the zip file https://github.com/RedBearLab/nRF51822-Arduino/archive/S130.zip
And unzip it and load the bootloader.hex to the mbed drive again to see.

sk1tt1sh, 2016-01-14:
Thank you. I managed to upload a custom built firmware (s120) on the nano. I will give this a whirl if I get a chance and let you know, but I learned how to write the code and merge it.

As a side note, I was unable to successfully do anything with windows. I had to use ubuntu vm with device filter to catch the MK20 plug in and the sudo cp -xv merged.hex /media/$USER/MBED/

Thanks for a fun product! :)

closed 2016-01-14 by sk1tt1sh