RedBearLab/nRF51822-Arduino #52: RBL 1.5 nano OTA not working, not showing up, not blinking

chrisallick, 2016-01-20:
Hello,
I just received a BLE Nano Kit in the mail. I followed the instructions: added the board via Arduino, using 1.6.7, selected the 1.5 nano from the boards, uploaded blink, nothing happened.

The light on the bottom of the nano was flashing for a while, but now it has stopped. The nrf toolbox app does not show the nano.

What's going on? Seemingly totally broken.

chrisallick, 2016-01-20:
Before it seemed to get through the "uploading" phase, now it simply just saying "Uploading..." forever.

Cheong2K, 2016-01-20:
Hi,
It is about hardware issue, please write to:
support@redbearlab.com

chrisallick, 2016-01-20:
The problem is that before anything will work with Arduino you MUST drag the new bootloader.hex file into the mBED drive. Then, after it reboots, the blink and BLEController sketch will work correctly.

Cheong2K, 2016-01-27:
Didn't you read this? https://github.com/RedBearLab/nRF51822-Arduino/blob/S130/README.md

closed 2016-01-27 by Cheong2K