RedBearLab/nRF51822-Arduino #55: BLE Central: launchServiceDiscovery doesn't discover all services and characteristics

therealchalz, 2016-02-01:
According to the documentation, if the Service and Characteristic UUIDs params are left to default (or are set to unknown) then the launchServiceDiscovery call should return ALL services and characteristics.  It does not do this.  The call seems to work if you know the Service and Characteristic UUIDs you are looking for (then it callbacks for exactly one service and one characteristic), but the code doesn't appear to make any attempt at looking for others when the UUIDs are set to unknown.

I noodled around but couldn't get the code in this repo to work properly.  I ended up pulling all the latest code from the following repos into the RBL one and massaging everything to work:
https://github.com/ARMmbed/ble.git
https://github.com/ARMmbed/ble-nrf51822.git
https://github.com/ARMmbed/nrf51-sdk.git

The BLE_CentralTest sample (still) works, and if the UUID params are omitted then the stack appropriately discovers all the available services and characteristics for each service.

Are there plans in the near future to update the nRF51822-Arduino code to the more recent mbed codebases?

therealchalz, 2016-02-02:
The code can be found [here ](https://github.com/therealchalz/nRF51822-Arduino) if anyone else wants to tinker.  Note that I was using the BLE NANO 32KB for most of the development.  People using other boards may/will have to update boards.txt with new locations for header files if they have compilation problems

zoujixing, 2016-06-08:
Hi theralchalz,
We will solve this problem at version 1.0.6 released this week.

gaucho1978, 2016-06-12:
I have similar problem: after the "ble.connect..." the connectionCallback is never fired. Will this be solved with 1.0.6 release or this is another problem? do i miss some line of code? I'm connecting a Polar H7 Heart Rate Sesor.