RedBearLab/nRF51822-Arduino #59: reflash mk20 not working

petrometro, 2016-03-19:
Hello- I wasn't able to get my MK20 to work with arduino 1.6.7, so I flashed it with the firmware found at the bottom of this page

https://developer.mbed.org/platforms/RedBearLab-BLE-Nano/

from there, I was able to get the MK20 to work via mbed's platform.

I decided to try again with Arduino using your instructions in the MK20 folder, and everything is as described EXCEPT for the last step.

The led flashes rapidly, I disconnect the MK20, and on reconnecting... no MBED directory and the led does not come on.

I am able to reflash with the mbed firmware each time and it works as expected.

I'm more familiar with Arduino and it's libraries than I am with mbed, so I'd really prefer Arduino.
Any ideas ?

I'm on macos but I tried the same steps with a windows 8 machine and had exactly the same results

Cheong2K, 2016-03-22:
Did you load the bootloader to the BLE Nano via the MK20 dongle?

petrometro, 2016-03-23:
yes, this was done as explained on the github readme

Cheong2K, 2016-03-24:
Did you install the driver and select the correct COM port?

petrometro, 2016-03-24:
as i said in my original issue, i'm on mac os so no driver to install (according to your docs), and i'm using the port that shows up in arduino when i plug the mk20 into my usb port.

Cheong2K, 2016-03-25:
If you can drag and drop the bootloader.hex to the BLE Nano but cannot upload sketch from the Arduino IDE, it may be the problem that the TX/RX pins of the serial port was not soldered very well, please check it. Also, take a screen dump of the IDE log for evaluation.

snicolai, 2016-05-24:
I'm on a mac with 10.11.5 and Arduino 1.6.9 and I've gotten the boot loader from the S130 branch and the board support package 1.0.4 to upload successfully.  1.0.5 gave me errors during upload on an older 2007 mac, but 1.0.5 works on late 2014 laptop.  1.0.4 worked on both.

If you use the boot loader from the trunk, it has the S110 soft device and requires the 1.0.0 board support package.  I've had that working as well.

tusharvb19, 2018-01-13:
using MK20 V1.0 but its not detecting in W10, ubuntu 16.04 after I upgrade the fw bootloader..I followed steps
The MK20 is going into bootloader mode by pressing reset but it is not detecting as MBED or DAPLINK and led is off.
Need help.