RedBearLab/nRF51822-Arduino #60: Indefinite upload, lacking programmer?

stuper27, 2016-03-27:
I went ahead and copied the url into preferences and then used the board manager to get BLE nano board down. Next step, so it says, is to compile and upload. Everything is fine up until I attempt to upload to the board. It just hangs forever... Any thoughts? The programmer currently selected is the AVRISP mkII which I am fairly sure that is incorrect, though there is no redbear programmer present as I have seen indicated in other issues I have read on here.

Cheong2K, 2016-03-28:
Did you load the bootloader.hex yet?

emcniece, 2016-04-19:
Just hit this as well - can confirm that the instructions at http://redbearlab.com/getting-started-nrf51822 leave this step out.

1. Place board on USB backpack thing, plug in to computer (file explorer should open automatically on Windows)
1. Download [repo zip](https://github.com/RedBearLab/nRF51822-Arduino), extract `bootloader/bootloader.hex`
1. Drag `bootloader.hex` in to file explorer folder (device will auto-flash)
1. Try using Arduino again

After doing this, the Arduino IDE no longer hangs at "Uploading...". Instead, it shows this:

```
Sketch uses 5,916 bytes (3%) of program storage space. Maximum is 196,608 bytes.

Found programmer: Id = "RBL-nRF"; type = S
Programmer supports buffered memory access with buffersize=256 bytes.
```

My system stats:

* Win8.1
* Arduino IDE 1.6.5
    * Board: BLE Nano(V1.5 32KB)
    * Port: COM5 (in Device Manager, reads as "mbed Serial Port (COM5)"
    * Programmer: "AVRISP mkII"

I am uploading the Examples > 0.1Basics > Blink example, modified with a `Serial` print statement - the serial monitor works too!

CCommander, 2016-06-28:
I would like to follow the instructions above, but my bootloader drive just has 124 KB and the bootloader.hex is much bigger. Is something wrong with my hardware?

I assume, I have the Board Version 1.0 isn't that supported anymore?

Cheong2K, 2016-06-29:
With the latest board package, you can burn the bootloader within the Arduino IDE, also, please download the RAW file of the bootloader.hex if you need to use the drag-&-drop method.

CCommander, 2016-07-01:
I tried to do that, but the shown flash drive in my explorer is smaller than the file size. Thus it cannot be dragged and dropped there.