RedBearLab/nRF51822-Arduino #63: SCAN REPORT don't decode the full name

gaucho1978, 2016-05-28:
on nRF51822-Arduino repository there is a bug in the scan callback.

In the "scan as master role" example of this repository you use "ble.startScan(scanCallBack)" to start the scanning.
the result data is obtained in the array "const Gap::AdvertisementCallbackParams_t *params"

if you use a heart rate sensor you will find that the raw inquiry response of the GAT is bigger than 18 bytes (36bytes in my heart rate sensor), but the length of the obtained data is only 18 bytes.

running the scanner example and adding some prints to the "ble_advdata_parser" function,  you will find that you will be able to decode the inquiry response correctly, but the last field is not included in the raw response data, so the loop on the "ble_advdata_parser" function will not be able to decode the last field. Last field in my case is the name of the bluetooth peripherial (Hearth rate sensor). 
You are cutting somewhere the bluetooth raw device response.
I verified the raw device response with a tablet and a Nordic app.

i would like to to some more research but may be you can solve faster the problem knowing better than me the source code.

gaucho1978, 2016-05-28:
may be it is a problem with softdevice S130

gaucho1978, 2016-05-29:
I used the heart rate sensor manufactured by POLAR  model H7.

The used example is https://github.com/RedBearLab/nRF51822-Arduino/blob/S130/arduino-1.6.x/hardware/RBL/RBL_nRF51822/libraries/BLE_Examples/examples/BLE_CentralTest/BLE_CentralTest.ino where i just added some prints to ble_advdata_parser subroutine.

gaucho1978, 2016-06-03:
Please help me.

zoujixing, 2016-06-06:
Hi gaucho,
I not sure what you mean. The advertisement data has 36 bytes?

gaucho1978, 2016-06-06:
Yes

gaucho1978, 2016-06-06:
I tried to use the new nordic sdk, but I had a lot of compile errors and at the end i wasn't able to solve the problem

Did you ever tried to make a scan with a heart rate sensor around? I found that the raw received advertisement data is 18 bytes but repeating the scan with a android app on my tablet I can see that the raw advertisement data has 36 bytes and from byte 19 to 36 there is the name of the device.

For this reason your scan example is not able to find the name of the scanned heart rate sensor.

gaucho1978, 2016-06-06:
I also tried to increase the size of some arrays but it didn't solved the problem, but I wasn'the able to find the source of the advertisement data.

For sure I think that you should find a method to simulate the heart rate sensor, if you don'the have one.

Can I help you in any way?

If you need it, I can send you here the raw data advertised by my device

zoujixing, 2016-06-06:
Hi gaucho,
It's not possible, the advertisement not more than 31bytes.  Maybe the name field belong to scan response data. On some app, we can see advertisement and scanRsp data at same time.

gaucho1978, 2016-06-06:
This is the raw data that you can read with the nrf app for android named nRf master controller:
0x02010603020D1807FF6B00371C4444020A001209506F6C6172204837203837334237433137
I copied this data from the Scanner tab. Just start the scan, single click on the found device then press the "raw" button and you will see the entire gatt advertised data. You will also find there the following link to help users to decode the advertised data: https://www.bluetooth.org/en-us/specification/assigned-numbers/generic-access-profile

gaucho1978, 2016-06-06:
The data that your example is able to get is just the first 18 bytes of the complete advertised data: 
0x02010603020D1807FF6B00371C4444020A00

gaucho1978, 2016-06-06:
Are you sure of what you wrote?

gaucho1978, 2016-06-06:
As you can see it is missing the field  «Complete Local Name» (0x09) and it's length is 18 (0x12)

gaucho1978, 2016-06-06:
How can I get the complete scan response data, included the "complete Local Name"?

gaucho1978, 2016-06-06:
You were right. Using your hint I found a function to enable the active scanning.
I added to the example the following line: ble.setActiveScan (true) 
Now it works
Please add this line to the example and a comment to explain the usage because otherwise, without any documentation, the users loose a lot of time behind this.
Thank you.

zoujixing, 2016-06-07:
Hi gaucho,
Good point! We will add this to next version.Thanks very much!

closed 2016-06-07 by zoujixing

straccio, 2016-11-08:
I have this problem too... the example have "setActiveScan(true)