RedBearLab/nRF51822-Arduino #68: Cannot upload sketch.

asukiaaa, 2016-07-16:
Hi, I tried to upload blink sketch to BLEnano but failed.
```
Sketch uses 6,012 bytes (2%) of program storage space. Maximum is 262,144 bytes.
Open On-Chip Debugger 0.9.0-gd4b7679 (2014-10-03-00:26)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
cortex_m reset_config sysresetreq
adapter speed: 1000 kHz
Error: unable to find CMSIS-DAP device

the selected serial port 
 does not exist or your board is not connected
```
Is there any way to solve this problem about openOCD?

My environment.

| Tool name | Version |
| --- | --- |
| OS | Ubuntu16.04 |
| ArduinoIDE | 1.6.9 |
| RedBearLab nRF51822 Borards | 1.0.6 |
| MK20 | 1.5 |

Thanks.

kdsoo, 2016-07-18:
I'm facing same problem and tried to roll back to bsp version 1.0.5 which works for now.

Cheong2K, 2016-07-18:
It seems that on Linux, there is some issues with the 1.0.6 board package, thanks for pointing out this, we are looking into this now.

sfoster, 2016-07-19:
I'm on Ubuntu 14.04 and I was experiencing this problem with 1.0.5 in the current Arduino IDE (1.6.8). Rolling the board version back to 1.0.4 "fixed" the issue for me.

asukiaaa, 2016-07-19:
Thanks for information!
I succeeded in uploading sketch with 1.0.5 board package on IDE 1.6.9.
I hope that we can upload sketches with the latest version board package in future.

closed 2016-07-19 by asukiaaa