RedBearLab/nRF51822-Arduino #69: Sleep and wakeup from external interrupt

kdsoo, 2016-07-20:
I'm making a prototype of wearable BLE gadget with ble nano.
To minimize power consumption, I'd like to leverage sleep API to make conditional sleep.
I can't find any from provided examples, is there any out of box sleep and wakeup API?
If wake up from external interrupt is supported, that will be the best for me.

kdsoo, 2016-07-25:
If ble nano does not support proper sleep and wakeup feature, there is no point to make the board in such a small size.
Does anyone has tried this successfully?

llemtt, 2016-08-10:
You should take a look at https://gist.github.com/sandeepmistry/6a6dfe9021b19bfddd60

sd_app_evt_wait() sleeps and waits for an event

kdsoo, 2016-08-11:
Thanks @llemtt .

It seems to be a nordic api isn't it?
So not whole functionalities of the BLE firmware are exported as Arduino library functions.
From now if I don't find proper API in Adruino library, I'll check nordic API as well.

closed 2016-08-11 by kdsoo