RedBearLab/nRF51822-Arduino #70: Rename Nordic S130 header back to ble.h (instead of nrf_ble.h)

sandeepmistry, 2016-07-25:
Hi @zoujixing,

I've noticed the in the v1.0.6 release the S130 ble.h header was renamed to nrf_ble.h (https://github.com/RedBearLab/nRF51822-Arduino/commit/b18d721da79ed784f54dd8a64a5b693535555568). Do you know why this was done?

(This is the file in `arduino-1.6.x/hardware/RBL/RBL_nRF51822/cores/RBL_nRF51822/ble-nrf51822-master/source/nordic_sdk/components/softdevice/s130/headers/nrf_ble.h`).

It's causing some incompatibilities with my [arduino-BLEPeripheral library](https://github.com/sandeepmistry/arduino-BLEPeripheral) see https://github.com/sandeepmistry/arduino-BLEPeripheral/issues/70 for more details. Can it be renamed back to `ble.h` in the v1.0.7 release?

cc/ @tigoe @szotyi007 @punamdahiya

zoujixing, 2016-07-25:
Hi sandeepmistry,
    Thanks for your point.We will fix it in version1.0.7.

tigoe, 2016-07-25:
Thanks, all, this will be a big help I imagine.

sandeepmistry, 2016-07-25:
@zoujixing excellent, thank you for making this change! Looking forward to the v1.0.7 release.

closed 2016-07-27 by zoujixing