RedBearLab/nRF51822-Arduino #73: openocd issue on Linux

kdsoo, 2016-08-11:
Hi,

I recently upgraded to bsp version 1.0.7 which seems to be the latest version and currently having problem just like we had in version 1.0.6. (revisiting.... #68 )

I rolled back to version 1.0.5 for now, which has no openocd problem.

But rolling back to 1.0.5 has also some problem. It won't compile some example codes with API function which have changed arguments 

Please let me know if there is solution for this.

kdsoo, 2016-08-11:
`Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-09-12:07)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : only one transport option; autoselect 'swd'
cortex_m reset_config sysresetreq
adapter speed: 1000 kHz
Error: unable to open CMSIS-DAP device 0xd28:0x204

the selected serial port 
 does not exist or your board is not connected`

This is the error message I got with version 1.0.7. 
openocd version is different from the one of 1.0.5 but still not working

Cheong2K, 2016-08-12:
Thank you for pointing out this, we will look into this.

zoujixing, 2016-08-16:
Tye open Arduino with sudo privileges

kdsoo, 2016-08-17:
@zoujixing ,
Does that really resolve the problem? if the running privilege is the issue, I think "setcap" could solve the problem.
anyway I tried "sudo ./arduino" but the most irritating thing launching ide with sudo is the whole workspace and libraries are blanked because of the privilege changed to root (home directory get changed)
It seems to be solving the openocd problem but I don't think this is the solution.
running as root privilege is the last thing that I want to do on my Linux box...

kdsoo, 2016-08-17:
I tried setcap on openocd (sudo setcap cap_net_raw+eip PATH_TO_REDBEAR_OPENOCD/openocd)
but didn't go well.
since launching arduino ide with sudo doesn't invoke openocd problem, privilege seems to be the cause but can't figure it out further.

bapowell, 2016-09-04:
I had the same problem (under Ubuntu 14.04). After much googling, I found this: [openocd.udev](http://openocd.zylin.com/#/c/1542/17/contrib/openocd.udev), which shows the following udev rules for the CMSIS-DAP device:
```
# mbed CMSIS-DAP
ATTRS{idVendor}=="0d28", ATTRS{idProduct}=="0204", MODE="664", GROUP="plugdev"
KERNEL=="hidraw*", ATTRS{idVendor}=="0d28", ATTRS{idProduct}=="0204", MODE="664", GROUP="plugdev"
```
After ensuring that my userid was a member of the plugdev group, I created a new udev rules file, /etc/udev/rules.d/98-blenano.rules, with these rules; then reloaded with `$ sudo udevadm control --reload-rules`, plugged in the MK20 device, and it worked.

kdsoo, 2016-09-04:
@bapowell Thank you. It works like a charm.
I couldn't imagine that mk20 is the problem since it's still ttyACMx.
Anyway I think the solution should be described in readme to avoid confusion.

closed 2016-09-04 by kdsoo

asukiaaa, 2017-01-23:
Thanks for good information.
I succeeded with the bapowell way.

Arduino IDE supports its rules so I also succeeded with [arduino/OpenOCD/contrib/60-openocd.rules](https://github.com/arduino/OpenOCD/blob/master/contrib/60-openocd.rules)

I set the rules by the following commands.
```sh
cd /etc/udev/rules.d
sudo wget https://raw.githubusercontent.com/arduino/OpenOCD/master/contrib/60-openocd.rules
sudo udevadm control --reload-rules
```

Thanks.

gremi64, 2024-12-05:
I had a similar problem with a Arduino Nano Matter (ubuntu 22.04/ArduinoIde2).

Thank you @asukiaaa, it works !