RedBearLab/nRF51822-Arduino #80: Compiled sketches don't work with just a softdevice

shlomozippel, 2016-10-15:
I used to be able to compile sketches with the RBL core and upload them to nrf51 based boards using external programmers / nrf51 dev kit and they would work fine, provided I also flashed the s130 softdevice. This was my workflow with version 1.0.5 of the package.

I have been trying to do the same with version 1.0.7 and couldn't get it to work.

If I upload the softdevice then my user code doesn't seem to get called, and if I upload the bootloader.hex that is part of the release then the device boots into the bootloader but never reaches the application code.

Is it possible to either change the bootloader to be able to load application code that wasn't uploaded through the bootloader, or alternatively have sketches using the RBL core work with the standard nordic softdevice without relying on the bootloader hex?

Thanks!

rogerclarkmelbourne, 2016-10-15:
Which soft device did you try using.

You need to use the S130 SD 1.0 from V10 of the SDK.

You can't use the SDK11 or SDK12 SD's as they are V2.0 and have a different code location address.
I suspect also that SDK 11 (or 12) may have changed some things like the number of arguments passed to functions etc, so as this repo uses SDK10 you can't simply change the linker file to the new application code start address and use SD2.0

You also have to use the S130 SD, as this effects the application code start address

BTW.
Really RBL's bootloader.hex, is just SD130 (V1.0) + OTA bootloader, so if you don't want the bootloader (which I don't either), I just use S130 V1.0 on its own

shlomozippel, 2016-10-15:
Thanks @rogerclarkmelbourne, that was indeed the problem. Just flashed s130 v1 and it works!

Great idea re: static library, I'll do that too

rogerclarkmelbourne, 2016-10-15:
@shlomozippel 

My version with the static lib is here

https://github.com/rogerclarkmelbourne/Arduino_nrf51822

But I have removed the RBL boards, as my primary focus is on generic nRF51 boards and iBeacons etc and smart watches

So you'd need to add whichever RBL board you are using , back into the boards.txt etc of my repo

Also, mine is likely to be a bit out of date as I can't simply pull from this repo, as I had to refactor the files into the system folder to make the static lib

BTW.
I did  video showing the repo being used with a smartwatch https://www.youtube.com/watch?v=SBHyQ1OSM-4

You could also take a look at Sandeep Mistry's repo https://github.com/sandeepmistry/arduino-nRF5
As this is a cleaner implementation than the RBL one, as it doesnt have all the mbed legacy code hiding under the surface

But I don't know what RBL boards it supports

shlomozippel, 2016-10-15:
Great stuff, thanks. I'm not actually targeting any RBL boards, I'm using PTR5518 modules and just using the RBL core as a starting point

rogerclarkmelbourne, 2016-10-16:
OK

I'd not seen those modules before, but they look somewhat overpriced on AliExpress

I guess the high price is because its CE certified.

closed 2016-10-17 by shlomozippel