RedBearLab/nRF51822-Arduino #82: Wire.begin() affecting clock based functions

dbradby, 2016-10-26:
When I used Wire.begin in my setup() method, both delay(...) and millis() don't work correctly. 

delay() blocks forever and millis() returns 0

If I comment out the Wire.begin() both those functions work again. Small sample code below ...
--------------
#include <Wire.h>

unsigned long myTime;

void setup() {
  Serial.begin(9600);
  
  Serial.println("starting setup");
  pinMode(13, OUTPUT);
  Wire.begin();
  Serial.println("setup complete");
}

void loop() {
  Serial.println("start loop");
  digitalWrite(13, HIGH);
  delay(1000);              
  digitalWrite(13, LOW);   
  delay(1000);
  myTime = millis();
  Serial.println(myTime);
  Serial.println("end loop");              
}

zoujixing, 2016-10-28:
Thanks for your point, I will test this.

zoujixing, 2016-10-31:
I have test the sample code, but I don't meet the problem.It works fine!
Here is some log:
```
start loop
349059
end loop
start loop
351088
end loop
start loop
353118
end loop
start loop
355148
end loop
start loop
357178
end loop

```

dbradby, 2016-11-19:
Thanks for confirming that it can work. I went back and looked at the board I was selecting. I previously had selected "BLE Nano v1.5" so instead I tried RBL nRF51822 v1.5) and it works!

What is the difference between the two boards and why should I select one over the other?

dbradby, 2016-11-20:
So I dug a little deeper myself and I can see the difference between the two is PIN assignments. 

The RBL nRF51822 profile uses D14 and D15 by default
BLE Nano uses D2 and D3 which is what I want to be using. 
When I switch to using the D2 and D3 pins then I see the delay and millis methods stop working and blocking. 

If I had to guess, it looks like the SCL assignment is hijacking the clock from the general MCU functions.

zoujixing, 2016-11-22:
So your board is RBL nRF51288 V1.5. I use the following code to test:
```
#include <Wire.h>

unsigned long myTime;

void setup() {
  Serial.begin(9600);

  Serial.println("starting setup");
  pinMode(13, OUTPUT);
  Wire.begin(D3, D2, TWI_FREQUENCY_100K);
  Serial.println("setup complete");
}

void loop() {
  Serial.println("start loop");
  digitalWrite(13, HIGH);
  delay(1000);              
  digitalWrite(13, LOW);   
  delay(1000);
  myTime = millis();
  Serial.println(myTime);
  Serial.println("end loop");              
}
```
I have used the D3/D2 for SCL/SDA,  I still didn't meet the problem.It works fine!
Can you tell me what you have done?