RedBearLab/nRF51822-Arduino #85: SCAN REPORT don't decode the full name

straccio, 2016-11-09:
With 1.0.7 version SCAN REPORT example don't decode the full name

https://github.com/RedBearLab/nRF51822-Arduino/issues/63

zoujixing, 2016-11-09:
Can you post you code here? I will check that.

straccio, 2016-11-09:
I have used the BLE_ScanReport example.

straccio, 2016-11-09:
I'm not sure, but before update the bootloader i remember that the code works

zoujixing, 2016-11-09:
Can you post some debug info?

straccio, 2016-11-09:
I have attached the code 
[BLE_ScanReport_FULLINFO.txt](https://github.com/RedBearLab/nRF51822-Arduino/files/580335/BLE_ScanReport_FULLINFO.txt)
and here some print to serial
`
Scan Device CallBack Handle 
  The peerAddr : 91 47 53 35 38 84  
  The Rssi : -49
  params->isScanResponse: 0
  params->type: 0
  params->advertisingDataLen: 11
  params->advertisingData: 2 1 6 7 FF 4C 0 10 2 B 0  
 
Scan Device CallBack Handle 
  The peerAddr : 91 47 53 35 38 84  
  The Rssi : -48
  params->isScanResponse: 1
  params->type: 0
  params->advertisingDataLen: 0
  params->advertisingData:  
 
Scan Device CallBack Handle 
  The peerAddr : 73 DD B0 BF 9C 70  
  The Rssi : -49
  params->isScanResponse: 0
  params->type: 0
  params->advertisingDataLen: 23
  params->advertisingData: 2 1 6 13 FF 4C 0 C E 8 4D 17 CA CB 67 D8 AB 2D 3A EB BF 1D F0  
 
Scan Device CallBack Handle 
  The peerAddr : 73 DD B0 BF 9C 70  
  The Rssi : -48
  params->isScanResponse: 1
  params->type: 0
  params->advertisingDataLen: 0
  params->advertisingData:  
 
Scan Device CallBack Handle 
  The peerAddr : 91 47 53 35 38 84  
  The Rssi : -49
  params->isScanResponse: 0
  params->type: 0
  params->advertisingDataLen: 11
  params->advertisingData: 2 1 6 7 FF 4C 0 10 2 B 0  
 
Scan Device CallBack Handle 
  The peerAddr : 73 DD B0 BF 9C 70  
  The Rssi : -50
  params->isScanResponse: 1
  params->type: 0
  params->advertisingDataLen: 0
  params->advertisingData:  
 
Scan Device CallBack Handle 
  The peerAddr : 73 DD B0 BF 9C 70  
  The Rssi : -52
  params->isScanResponse: 0
  params->type: 0
  params->advertisingDataLen: 23
  params->advertisingData: 2 1 6 13 FF 4C 0 C E 8 4D 17 CA CB 67 D8 AB 2D 3A EB BF 1D F0`

zoujixing, 2016-11-09:
Your advertisement don't contain a name type(BLE_GAP_AD_TYPE_SHORT_LOCAL_NAME: 0x08, BLE_GAP_AD_TYPE_COMPLETE_LOCAL_NAME: 0x09)

straccio, 2016-11-09:
I think there's a problem, from a bluetooth browser i can see other devices like a magic mouse or an iPhone with name. These devices are near (2 or 3 cm) from the blenano.
I seem that the peer address is inverted.

straccio, 2016-11-09:
I can confirm that the peer address is inverted.
I'm not an expert of Bluetooth but is normal that the scanresponse have no advertisingData?

zoujixing, 2016-11-09:
That is device name not include in advertisement and scan response data.