RedBearLab/nRF51822-Arduino #87: Cannot burn bootloader

venusyu0713, 2016-11-13:
To burn bootloader, I try to drag the bootloader.hex file into driver(MBED(E:))
But it returns error. It means "bootloader.hex file is too big, It cannot use in file system" (sorry my system language is not  English)

How can I burn bootloader?

zoujixing, 2016-11-14:
How much the driver(MBED)'s capacity?

venusyu0713, 2016-11-15:
Driver total capacity is 1.01MB and available capacity is 0.99MB. 
bootloader.hex file size is 337KB.

I use RBL nRF51822 v1.5 board and OS is windows 10.

zoujixing, 2016-11-15:
That is right capacity.
Try to use cmd.exe to load bootloader.hex( copy bootloader.hex E: ).

venusyu0713, 2016-11-15:
Thank you! It works :)

closed 2016-11-15 by venusyu0713

tusharvb19, 2018-01-13:
using MK20 V1.0 but its not detecting in W10, ubuntu 16.04 after I upgrade the fw bootloader..I followed steps
The MK20 is going into bootloader mode by pressing reset but it is not detecting as MBED or DAPLINK and led is off.
Need help.