RedBearLab/nRF51822-Arduino #88: MBED driver is connected & disconnected continuously

venusyu0713, 2016-11-16:
When RBL nRF51822 board connect with PC, MBED driver (E:\) is connected and disconnected continuously.
Power isn't off but BLE signal is off.

I tested it another PC but same problem occured.
Is it driver problem??

zoujixing, 2016-11-16:
Try to update the interface firmware to DAPLink.
[https://github.com/RedBearLab/nRF51822-Arduino/tree/S130/interface/mk20/board_v1.0/daplink](url)

venusyu0713, 2016-11-16:
How to update interface firmware??

zoujixing, 2016-11-16:
Press button, then connect board to PC, you will see a driver "Bootloader".
Drag the bin file into the driver.
By the way, what is your PC system?

venusyu0713, 2016-11-16:
Windows 10.

After update the interface firmware, my pc doesn't detect the driver(MBED).
Bootloader is detected when I plug board and then push the reset button.

zoujixing, 2016-11-16:
what is your board version? 1.0?

venusyu0713, 2016-11-16:
nRF51822 V1.5 (32KB)

zoujixing, 2016-11-16:
Sorry, nRF51822 V1.5 can't use the new interface firmware, just use old [https://github.com/RedBearLab/nRF51822-Arduino/tree/S130/interface/mk20/board_v1.5/cmsis-dap](url)

Do you have other PC with window7 or linux? We meet this problem on window10, but it should work on other system.

venusyu0713, 2016-11-18:
On Ubuntu system, It works. Thanks.

closed 2016-11-18 by venusyu0713

tusharvb19, 2018-01-13:
using MK20 V1.0 but its not detecting in W10, ubuntu 16.04 after I upgrade the fw bootloader..I followed  steps
The MK20 is going into bootloader mode by pressing reset but it is not detecting as MBED or DAPLINK and led is off.
Need help.