RedBearLab/nRF51822-Arduino #96: Serial 9600 baud limitation

hieuvt6-fti, 2016-12-21:
Hi,

I have been using the nRF51822 and I am interested in trying out the nRF51822 Arduino project.
One thing strikes me as I read the README is the limitation of 9600 bps while using BLE.

As far as my experience with the nRF51 series goes, I have never had any problem running UART at baud rate as high as 460800, never with flow control. That is done Nordic's BLE Stack already in their SoftDevice.

Could we maybe review this and find an improvement?

zoujixing, 2017-01-12:
Hi hieuvt6-fti,
We have done a test that when the baudRate exceeds 9600,  maybe lose some bytes (Uninterrupted transmission). BaudRate 9600 is the most stable.

khertan, 2017-03-03:
Ouch .... 9600 this hurts ... This explain why i lose bytes at 38400 ...
So the only solution is to avoid arduino and use Nordic SDK ? Or is there some things to avoid that could limit such loss ?